ARMmbed/mbed-crypto #21: psa_key_agreement API limitation

nazrchorn, 2019-01-19:
### Description
psa_key_agreement API doesn't have salt and label parameters. 
It calls  psa_key_derivation_internal with NULL as a salt and label.

Lets add four following parameters to psa_key_agreement API:
const uint8_t *salt, size_t salt_length,
const uint8_t *label, size_t label_length
And use them for psa_key_derivation_internal function call.

### Issue request type
    [ ] Question
    [x] Enhancement
    [ ] Bug

gilles-peskine-arm, 2019-01-21:
Hi Nazar,

Thank you for your interest in the PSA crypto API. We're about to release an updated version of the API which resolves this limitation in a different way. We're splitting the call to set the salt and label from the call to pass the secret key or shared secret. This way you can pass all the inputs that the KDF requires regardless of whether the secret is calculated from a key agreement or not.

```
psa_crypto_generator_t derivation = {0};
psa_key_derivation_setup(&derivation, PSA_ALG_KEY_AGREEMENT(PSA_ALG_ECDH, PSA_ALG_HKDF(PSA_ALG_SHA_256));
psa_key_derivation_input_bytes(&derivation, PSA_KDF_STEP_SALT, salt, salt_length);
psa_key_agreement(&derivation, PSA_KDF_STEP_SECRET, our_key, their_key_bytes, their_key_length);
psa_key_derivation_input_bytes(&derivation, PSA_KDF_STEP_INFO, info, info_length);
psa_crypto_generator_output_key(&derivation, aes_key, PSA_KEY_TYPE_AES, 16);
psa_crypto_generator_abort(&derivation);
```

nazrchorn, 2019-01-23:
Looks good for me.
Please let me know when it will be released, so I will pick it up.

gilles-peskine-arm, 2019-09-23:
Hello,

The new API has been released in Mbed Crypto 2.0.0. Key derivation now takes inputs in separate steps as described above, so taking one input of a KDF from a key agreement is compatible with taking other inputs.

closed 2019-09-23 by gilles-peskine-arm