ARMmbed/mbed-crypto #93: Document the stability of PSA interfaces

gilles-peskine-arm, 2019-04-12:
### Description

As the PSA interfaces are stabilizing, we should document what we'll maintain backward compatibility for and what we won't.

* Public APIs that haven't completely stabilized yet, but we'll provide a short transition period if we make changes: `include/crypto/crypto.h`, `include/crypto/crypto_types.h`, `include/crypto/crypto_values.h` (only the macros that have Doxygen documentation), `include/crypto/crypto_extra.h`. We should also note that even if the interfaces stabilize, that doesn't mean that the implementation is production-quality yet.
* Internal definitions that are platform-dependent and may change without notice: `include/crypto/crypto_platform.h`, `include/crypto/crypto_struct.h`
* File storage for persistent keys
* Driver HAL: not stable yet.

Where to document it: in the comment at the top of each header, in `README.md`, in the Doxygen documentation, …

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2023-05-24:
The PSA interfaces have stabilized so this is moot.

closed 2023-05-24 by gilles-peskine-arm