ARMmbed/mbed-crypto #116: psa_asymmetric_verify failed to verify key pair

moshe-shahar, 2019-05-19:
_`mbedtls_pk_verify`_ for PSA failed to verify the key pair below.

Using EC key pair and hash below, mbedtls_pk_verify call to _psa_asymmetric_verify_ which return `MBEDTLS_ERR_ECP_VERIFY_FAILED`.
```
const uint8_t der_private_ec_key[138] = \{ 0x30, 0x81, 0x87, 0x02, 0x01, 0x00, 0x30, 0x13, 0x06, 0x07, 0x2a, 0x86, 0x48, 0xce, 0x3d, 0x02, 0x01, 0x06, 0x08, 0x2a, 0x86, 0x48, 0xce, 0x3d, 0x03, 0x01, 0x07, 0x04, 0x6d, 0x30, 0x6b, 0x02, 0x01, 0x01, 0x04, 0x20, 0x00, 0x03, 0xe1, 0xcc, 0x78, 0xee, 0x88, 0xe6, 0x00, 0x3c, 0x78, 0x60, 0xb4, 0x4d, 0x2e, 0x33, 0x99, 0x3b, 0x5e, 0x36, 0x55, 0xfa, 0xdf, 0x24, 0xe2, 0x33, 0xed, 0xf5, 0x2f, 0x13, 0x76, 0xe9, 0xa1, 0x44, 0x03, 0x42, 0x00, 0x04, 0xac, 0x36, 0x73, 0x98, 0x49, 0xb8, 0xe6, 0x99, 0xe9, 0x6a, 0xcd, 0xc8, 0x80, 0x50, 0xc4, 0xda, 0xeb, 0xd4, 0x09, 0xa0, 0xcf, 0x3f, 0x33, 0x46, 0x52, 0xa3, 0x0d, 0x29, 0x7a, 0x14, 0x6e, 0x17, 0x2a, 0x69, 0x19, 0x2e, 0xaf, 0x0a, 0x4e, 0xcf, 0xca, 0x5a, 0x10, 0xca, 0x67, 0x65, 0x57, 0x11, 0xd3, 0x48, 0x8c, 0xfb, 0xfc, 0x18, 0xf7, 0x3c, 0x72, 0x51, 0x02, 0x4b, 0x3d, 0x05, 0x9a, 0xe7 };
const uint8_t der_public_ec_key[91] = \{ 0x30, 0x59, 0x30, 0x13, 0x06, 0x07, 0x2a, 0x86, 0x48, 0xce, 0x3d, 0x02, 0x01, 0x06, 0x08, 0x2a, 0x86, 0x48, 0xce, 0x3d, 0x03, 0x01, 0x07, 0x03, 0x42, 0x00, 0x04, 0xac, 0x36, 0x73, 0x98, 0x49, 0xb8, 0xe6, 0x99, 0xe9, 0x6a, 0xcd, 0xc8, 0x80, 0x50, 0xc4, 0xda, 0xeb, 0xd4, 0x09, 0xa0, 0xcf, 0x3f, 0x33, 0x46, 0x52, 0xa3, 0x0d, 0x29, 0x7a, 0x14, 0x6e, 0x17, 0x2a, 0x69, 0x19, 0x2e, 0xaf, 0x0a, 0x4e, 0xcf, 0xca, 0x5a, 0x10, 0xca, 0x67, 0x65, 0x57, 0x11, 0xd3, 0x48, 0x8c, 0xfb, 0xfc, 0x18, 0xf7, 0x3c, 0x72, 0x51, 0x02, 0x4b, 0x3d, 0x05, 0x9a, 0xe7 };
uint8_t hash[32] = { 0xa0,0x63,0xdf,0x83,0xa8,0xc2,0x8a,0x49,0xda,0xf4,0xae,0xba,0x0e,0x29,0xee,0x7b,0x21,0x77,0xe8,0x51,0x10,0x72,0x94,0x4c,0x3d,0x29,0x9c,0xf7,0x7d,0xc8,0x3e,0x7a };
```

The non-PSA version succeeds to verify the same key pair.
Tested with [mbedtls version 2.17.0](https://github.com/ARMmbed/mbedtls/releases/tag/mbedtls-2.17.0).

shelib01, 2019-05-20:
@moshe-shahar , please print keys from mbedtls_pk_context (in both cases) before calling to  mbedtls_pk_verify and mbedtls_pk_sign and signatures that you receive and pass to  sign/verify functions .

moshe-shahar, 2019-05-20:
Sorry, the error was during the conversion of the key to raw format.

closed 2019-05-20 by moshe-shahar