ARMmbed/mbed-crypto #129: Port the windows mbedtls issue fix to mbedcrypto

KenLSoft, 2019-05-27:
Hi,
I am trying to compile mbedcrypto on windows, but meet the same problem while I was working for mbedtls.
I am using cmake, with git bash + cmake on windows, not a cygwin environment.

The error message is:

CMake Error at CMakeLists.txt:114 (message):
  Could not create symbolic link for:
  C:/Users/user01/git/mbed-crypto/include/mbedtls -->
Call Stack (most recent call first):
  include/CMakeLists.txt:20 (link_to_source)

I found the same issue at:
https://git.trustedfirmware.org/trusted-firmware-m.git/commit/?id=a0f505c659e2af2ffa278d215045add711f7f2d8

Shall we port the fix on mbedtls into this?

Thanks.

adeaarm, 2019-05-27:
Original issue that is being referred to is [available at this link](https://github.com/ARMmbed/mbedtls/issues/1496)

KenLSoft, 2019-05-27:
Thanks, let me close this one.

Duplicated with https://github.com/ARMmbed/mbed-crypto/issues/103

closed 2019-05-27 by KenLSoft