ARMmbed/mbed-crypto #134: Allow omitting OID name/description

pkolbus, 2019-06-02:
### Description

On small systems (e.g. Cortex-M4) every KB of code footprint counts, and the OID name and description string tend to only be needed in certain debugging/tools contexts (e.g. `mbedtls_x509_crt_info()`). I've noticed that when building with `-ffunction-sections`, `-fdata-sections`, and `gc-sections`, there are a few KB of wasted code space because constant strings are entangled to types in a way that makes it hard for a linker to omit them.

I have a similar approach in mind as https://github.com/ARMmbed/mbedtls/pull/2315, and propose the configuration `MBEDTLS_OID_NAMES`. If you agree I can get started on the PR.

Note that because mbed TLS now uses mbed-crypto, this issue is a dependency for that PR.

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2024-07-09:
This will finally be resolved through https://github.com/Mbed-TLS/mbedtls/issues/9379 in TF-PSA-Crypto 1.0 and Mbed TLS 4.0.

closed 2024-07-09 by gilles-peskine-arm