ARMmbed/mbed-crypto #135: Supply a Docker image for test execution

pkolbus, 2019-06-02:
### Description

The test scripts (especially `all.sh`) have a number of dependencies. Supplying a Dockerfile and helper scripts would make it easier for contributors (running Linux at least) to set up a complete environment for running tests.

https://github.com/ARMmbed/mbedtls/pull/2314 provides a good basis for this. 

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

hug-dev, 2019-12-22:
For future reference, as I came up to this issue when wanting to add a Mbed Crypto container for our own project, the following Dockerfile will install Mbed Crypto version `2.0.0` in `/tmp/` inside the container:

```
FROM ubuntu:latest

RUN apt-get update && \
    apt-get install -y make gcc python3 python wget

WORKDIR /tmp
RUN wget https://github.com/ARMmbed/mbed-crypto/archive/mbedcrypto-2.0.0.tar.gz
RUN tar xf mbedcrypto-2.0.0.tar.gz
RUN cd mbed-crypto-mbedcrypto-2.0.0 \
		&& make
```

gilles-peskine-arm, 2023-05-24:
The Docker images that the maintainer team uses are now publicly available in https://github.com/Mbed-TLS/mbedtls-test/tree/master/resources/docker_files

closed 2023-05-24 by gilles-peskine-arm