ARMmbed/mbed-crypto #137: different behavior in `calloc` on different libraries when trying to allocate 0 bytes

RonEld, 2019-06-04:
### Description

[man calloc])(https://linux.die.net/man/3/calloc) describes the following:
>If nmemb or size is 0, then calloc() returns either NULL, or a unique pointer value that can later be successfully passed to free().

This causes different behavior when calling the library wiht 0 bytes to allocate.
For example:
a call to  `psa_import_key()` will result in an error when `slen` is 0 because in `prepare_raw_data_slotoreoare_data_slot()` we will have:
~~~
    /* Allocate memory for the key */
    raw->bytes = PSA_BITS_TO_BYTES( bits );
    raw->data = mbedtls_calloc( 1, raw->bytes  );
    if( raw->data == NULL )
    {
        raw->bytes = 0;
        return( PSA_ERROR_INSUFFICIENT_MEMORY );
    }
~~~

and a call to `psa_key_derivation()` with the same empty key could result in a failure because in `psa_generator_tls12_prf_setup()`:
~~~
   tls12_prf->key = mbedtls_calloc( 1, key_len + 1 );
    if( tls12_prf->key == NULL )
        return( PSA_ERROR_INSUFFICIENT_MEMORY );
~~~

If using `memory_buffer_alloc` feature, the calloc calls would return `NULL` and not a valid pointer because:
~~~
    if( n == 0 || size == 0 || len / n != size )
        return( NULL );
~~~

There are several ways to fix this:
1. Don't call `mbedtls_calloc()` for a zero length -> This could cause crashes in the library for some cases where the data cannot be zero lengthed
2. Always add one more to the size to allocate -> This adds some additional RAM. One could argue this is insignificant.
3. Have the `memory_buffer_alloc()` return a 4 byte aligned pointer for zero length allocation request. This perhaps will fix the specific use case, but on other platforms, that have some calloc function that returns `NULL`, as it's allowed,the error will remain.
### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

hanno-arm, 2019-06-04:
FWIW, in https://github.com/ARMmbed/mbedtls/pull/2478 I use and argue for (2) of always adding a byte.

hanno-arm, 2019-06-04:
Alternatively, one can of course allocate `len ? len : 1` bytes.

gilles-peskine-arm, 2019-06-04:
0-length buffers with a null pointer are problematic for several reasons. One is that you can't check for malloc/calloc failure just by testing NULL. Another is that calling functions like memset and memcpy on NULL has undefined behavior even when the size is 0.

Regarding key sizes in particular: `prepare_raw_data_slot` has a special case for raw-data slots that handles size=0 correctly. Other key types don't have this special case because it doesn't make sense to have a 0-length _key_. It can make sense to use a raw-data slot with a size of 0, but it's such a rare need that I'm inclined to forbid it in order to simplify the implementation and have correct semantics in edge cases (in key attributes, 0 for a key size means “I don't know”). Forbidding 0-length keys is on the [list of planned changes before finalizing version 1.0 of the PSA crypto API](https://armmbed.github.io/mbed-crypto/html/index.html#planned-changes-for-version-1-0).

In any other place where we manipulate buffers that can have length 0, we need to be careful. It's not just (1) being careful with malloc, it's also being careful with memset and such.

(2) is not a solution because buffers can come from the caller and it's perfectly legitimate to pass NULL for a 0-size buffer. It may come from the platform's malloc, for example.

Incidentally @hanno-arm In general, adding 1 to the size doesn't add 1 byte of memory pressure on average, because data sizes are not distributed uniformly modulo the malloc block size: data with a size that's an exact multiple of the block size is very common, and in that case you get a whole block more. Conversely, adding 1 byte makes no difference if the data size is not a multiple of the block size, but that only compensates if the probability that a piece of data has the block size is exactly average, and it's usually more. For ASN.1-encoded data, the distribution is probably close to uniform, but for keys, it isn't.

(3) is not a solution because there are other platforms where malloc returns NULL, and because the application might decide to pass NULL for a 0-size buffer without calling malloc.

RonEld, 2019-06-04:
@gilles-peskine-arm 
1) Yes, this is what I meant by "This could cause crashes in the library for some cases where the data cannot be zero lengthed". Perhaps I didn't explain correct

2) This is why our suggetion is to add one to the size,so the platform's malloc will not return NULL, unless running out of memory.

3) Yes, this is why I mentioned "his perhaps will fix the specific use case, but on other platforms, that have some calloc function that returns NULL, as it's allowed,the error will remain." 

For these reasons I am in favor of (2)

> can make sense to use a raw-data slot with a size of 0, but it's such a rare need that I'm inclined to forbid it in order to simplify the implementation and have correct semantics in edge cases (in [...]

This will clash with [EAP-TLS](https://tools.ietf.org/html/rfc5216#section-2.3) key derivation functionality, since deriving the IV uses 0 length key

gilles-peskine-arm, 2019-06-04:
> This will clash with EAP-TLS key derivation functionality, since deriving the IV uses 0 length key

Oh. Thanks, I wasn't aware of that. I'm not convinced the best solution is to allow 0-length key objects though. Another way would be to allow the “key” input to be a buffer for this particular KDF (each KDF can have its own input sequence). Yet another way would be to define TLS-PRF-with-empty-key as a separate algorithm identifier.

Regarding (2), it only works where the library calls malloc internally. You still need to copy with 0-length buffers pretty much everywhere inside the library, so (2) has very little advantage.

gilles-peskine-arm, 2019-09-17:
* Fix in https://github.com/ARMmbed/mbed-crypto/pull/257
* Tests in https://github.com/ARMmbed/mbed-crypto/pull/264

PR #264 (2019-09-25): Test the library when malloc(0) returns NULL
On our main test platforms (Linux, FreeBSD), malloc(0) returns a non-NULL pointer. With `MBEDTLS_MEMORY_BUFFER_ALLOC_C`, `calloc(0,_)` does return NULL, but memory_buffer_alloc doesn't play well with UBSan. Add a test component with UBSan where `calloc` returns NULL when asked to allocate 0 bytes, wrapping around the platform `calloc`.

Fix #137: now it's tested.

Should be backported to mbedtls LTS branches.
closed 2019-09-25 by Patater