ARMmbed/mbed-crypto #169: Change unsigned char to uint8_t in the API.

steven-bellock, 2019-07-12:
### Description

Release 1.0 beta3 of the API Specification states that "A buffer parameter always has the type uint8_t * or const uint8_t *..." However there are six functions in the document

psa_cipher_generate_iv
psa_cipher_set_iv
psa_cipher_update
psa_aead_generate_nonce
psa_aead_set_nonce
psa_aead_update

that specify the buffer type as unsigned char, which is effectively equivalent to uint8_t, but it's good to be consistent.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

gilles-peskine-arm, 2019-07-12:
Thanks for the report! It turns out we had a fix floating around but it fell through the cracks during the last beta release of the specification. Now the fix is back on track. Thanks!

gilles-peskine-arm, 2019-07-15:
This is fixed on the branch which is used to generate the specification document. The fix will be included in the next release of the document.

PR #170 (2019-07-15): Use stdint.h types
Follow MISRA C 2012 rules by using exact width types from stdint.h.

Fix #169.
closed 2019-07-15 by gilles-peskine-arm