ARMmbed/mbed-crypto #171: Double-underscore in macro names is not valid in C++

gilles-peskine-arm, 2019-07-12:
### Description

Mbed Crypto headers define an identifier that is reserved in C++. Specifically `PSA__ALG_AEAD_WITH_DEFAULT_TAG_LENGTH__CASE` in `include/psa/crypto_values.h`. C++ reserves double underscores.

Note that it isn't enough to just change the double underscore to a single underscore: `test_psa_constant_names.py` picks up macros `PSA_ALG_.*` and would need to skip this one somehow. See https://github.com/ARMmbed/mbedtls-psa/pull/258 (private link).

Acceptance criteria:

* Add a check for double underscores. To `check-names.sh`, presumably. It already checks for trailing underscores, for whatever reason.
* Obviously, make the check pass without breaking anything else.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

AndrzejKurek, 2019-08-21:
PR created: https://github.com/ARMmbed/mbed-crypto/pull/230

gilles-peskine-arm, 2019-09-03:
Fixed by https://github.com/ARMmbed/mbed-crypto/pull/230

closed 2019-09-03 by gilles-peskine-arm