ARMmbed/mbed-crypto #175: psa_key_derivation_output_bytes

gowthamsiddarthd, 2019-07-16:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

athoelke, 2019-07-16:
Please include details of the key derivation algorithm you are using and the sequence of calls and step parameters to each call. The requirements for the steps and inputs depend on the algorithm.

gilles-peskine-arm, 2019-07-16:
`PSA_ERROR_BAD_STATE` sounds right. None of the existing key derivation algorithms can work with only a key.

gowthamsiddarthd, 2019-07-18:
Hi @athoelke @gilles-peskine-arm 
Below were the sequence of steps followed and the actual result

	1. Psa_crypto_init()
	2. If the input is PSA_KEY_DERIVATION_INPUT_SECRET, 
		a. Psa_set_key_type() with PSA_KEY_TYPE_DERIVE
		b. Psa_set_key_algorithm() as PSA_ALG_KEY_AGREEMENT(PSA_ALG_ECDH, PSA_ALG_HKDF(PSA_ALG_SHA_256)),
		c. Psa_set_key_usage_flags() with PSA_KEY_USAGE_DERIVE
		d. Psa_import_key() with 
			i. above attributes, 
			ii. Key data = {0x49, 0x8E, 0xC7, 0x7D, 0x01, 0x95, 0x0D, 0x94, 0x2C, 0x16, 0xA5, 0x3E, 0x99,
			 0x5F, 0xC9, 0x77}
			iii. Key size = 16
			iv. Key handle = (address of key_handle variable)
			v. Result = PSA_SUCCESS
	3. Psa_key_derivation_setup() with
		a. operation = PSA_KEY_DERIVATION_OPERATION_INIT
		b. Alg = PSA_ALG_KEY_AGREEMENT(PSA_ALG_ECDH, PSA_ALG_HKDF(PSA_ALG_SHA_256)),
		c. Result = PSA_SUCCESS
	4. Psa_key_derivation_set_capacity()
		a. Operation
		b. Capacity = 42
		c. Result = PSA_SUCCESS
	5. If the input is  PSA_KEY_DERIVATION_INPUT_SECRET,  call psa_key_derivation_input_key() with
		a. Operation
		b. Step = PSA_KEY_DERIVATION_INPUT_SECRET
		c. Key_handle
		d. Result = PSA_SUCCESS
	6. Else call psa_derivation_input_bytes() with
		a. Operation
		b. Step = PSA_KEY_DERIVATION_INPUT_INFO (Or salt, label or seed to cover all cases)
		c. Data = "This is the info"
		d. Size = 16
		e. Result = PSA_SUCCESS
	7. Call psa_key_derivation_output_bytes()
		a. Operation
		b. Output buffer
		c. Output length = 42
		d. Result = PSA_ERROR_BAD_STATE

gilles-peskine-arm, 2019-09-23:
With a key agreement algorithm, the `SECRET` step should fail. Looks like a bug. We don't have good test coverage for operation steps involving key agreement.

zaya-mc, 2020-05-12:
Hi all, what is the latest status in this issue? 

If we get a fail in the mentioned case, PSA Tests does not run the next cases. 
For example, in the second case, PSA Tests pass only "Info", does not pass a key and then calls "psa_key_derivation_output_key()". 
Without a source key, is it enough to derive a key using only "Info" because PSA Tests somehow expects Success. 
The other cases like Label also fail.

gilles-peskine-arm, 2021-05-27:
I wrote some unit tests in https://github.com/ARMmbed/mbedtls/pull/4576 and they didn't reveal any bugs in the current version of Mbed TLS. Is this issue still relevant?

Looking at https://github.com/ARMmbed/mbed-crypto/issues/175#issuecomment-512637268 again, I'm not sure what you're testing. Steps 2, 5 and 6 are listed as conditional on an input step, but that doesn't really make sense: a key derivation takes multiple input steps. Please let me know if there's a specific sequence of input steps that doesn't work the way you expect.

Summer-ARM, 2021-05-31:
could you add a test like this:

```
PSA key derivation: HKDF SHA-256, RFC5869 #1, output 42+0
depends_on:PSA_WANT_ALG_HKDF:PSA_WANT_ALG_SHA_256
derive_output:PSA_ALG_HKDF(PSA_ALG_SHA_256):PSA_KEY_DERIVATION_INPUT_SECRET:"0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b0b":""
```

only with step 'PSA_KEY_DERIVATION_INPUT_SECRET' and help to test ?

The issue is that: 

- in psa_key_derivation_ouput_bytes(), it will call psa_key_derivation_hkdf_read()
- in psa_key_derivation_hkdf_read(), it will check 'hkdf->info_set'
`if( hkdf->state < HKDF_STATE_KEYED || ! hkdf->info_set )
        return( PSA_ERROR_BAD_STATE );`
- however, in the whole operation, we only use step 'PSA_KEY_DERIVATION_INPUT_SECRET', so didn't set hkdf->info_set, that's why will return error.

gilles-peskine-arm, 2021-05-31:
I don't understand: did you mean to add a step `PSA_KEY_DERIVATION_INPUT_INFO` after this? `derive_output` only does positive test cases, and the info step is mandatory for HKDF.

Summer-ARM, 2021-05-31:
Thanks for your nice explanation, I got it! info step is mandatory for HKDF.

gilles-peskine-arm, 2021-05-31:
It appears that in addition to missing bad-state checks that have been fixed in Mbed TLS since this issue was raised, the compliance tests were incorrectly requiring INFO to be optional for HKDF. Since all related bugs in Mbed TLS appear to have been fixed, I am closing this issue. Please open a new issue in the mbedtls project if more bugs appear.

closed 2021-05-31 by gilles-peskine-arm