ARMmbed/mbed-crypto #176: psa_key_derivation_input_bytes with LABEL or SAL

gowthamsiddarthd, 2019-07-16:
### Description
		a. API: psa_key_derivation_input_bytes with label/seed as step
		b. Calling key derivation setup with alg such that PSA_ALG_IS_KEY_DERIVATION is true. Then, calling psa_key_derivation_input_bytes with step as PSA_KEY_DERIVATION_INPUT_LABEL. The return value is expected to be PSA_SUCCESS but the actual return value is PSA_ERROR_INVALID_ARGUMENT.
		The same case with step as PSA_KEY_DERIVATION_INPUT_SEED

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2019-07-16:
The permitted input depend on the algorithm. What algorithm did you pass to `psa_key_derivation_setup`? Did it succeed? Which version did you use (there's no tag `psa-crypto-api-1.0-beta`)?

gowthamsiddarthd, 2019-07-18:
@gilles-peskine-arm,
Below were the sequence of steps followed and the actual result
	1. Psa_crypto_init()
	2. Psa_key_derivation_setup() with
		a. operation = PSA_KEY_DERIVATION_OPERATION_INIT
		b. Alg = PSA_ALG_KEY_AGREEMENT(PSA_ALG_ECDH, PSA_ALG_HKDF(PSA_ALG_SHA_256)),
		c. The result is PSA_SUCCESS
	3. Psa_key_derivation_input_bytes() with
		a. Operation
		b. Step = PSA_KEY_DERIVATION_INPUT_LABEL
		c. Data = "abcdefghijklmnop"
		d. Size = 16
		e. The result is PSA_ERROR_INVALID_ARGUMENT

Patater, 2019-07-18:
This is a question about the API. There may be a bug here, but we don't know yet.

gilles-peskine-arm, 2019-09-23:
This is expected: HKDF does not take a `LABEL` or `SEED` input (it takes an optional `SALT`, then a `SECRET`, then an `INFO`).

closed 2019-09-23 by gilles-peskine-arm