ARMmbed/mbed-crypto #194: Spurious files and make rules in tests/data_files

gilles-peskine-arm, 2019-07-30:
### Description

Out of the box as of Mbed Crypto 1.1.1:
```
$ cd tests/data_files
$ mbedtls_cert_req output_file=test-ca.req.sha256 filename=test-ca.key password=PolarSSLTest subject_name="C=NL,O=PolarSSL,CN=PolarSSL Test CA" md=SHA256
/bin/sh: 1: mbedtls_cert_req: not found
Makefile:47: recipe for target 'test-ca.req.sha256' failed
make: *** [test-ca.req.sha256] Error 127
```

It isn't a big deal, but we should clean up the things in `data_files` that require X509.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

Patater, 2019-07-31:
For that Makefile, Mbed Crypto currently requires Mbed TLS to have been `make install`'d on your system in order to have the `mbedtls_cert_req` tool available. Alternatively, you can specify a path to the binary itself, if you've only done a `make` on your Mbed TLS copy and don't want to add it to your path.

Mbed Crypto also requires `openssl` for that Makefile.

gilles-peskine-arm, 2020-09-10:
No longer applicable since Mbed Crypto was merged back into Mbed TLS.

closed 2020-09-10 by gilles-peskine-arm