ARMmbed/mbed-crypto #203: Test "nonexistent" in psa its suite causes NULL p_data to propagate to fwrite in psa_its_set

nanokatze, 2019-08-06:
# Description

Mbed Crypto 1.1.1
Toolchain GCC 9.1.1
Target Linux 5.1.20 x86_64 (Fedora 30 5.1.20-300)

Test
https://github.com/ARMmbed/mbed-crypto/blob/mbedcrypto-1.1.1/tests/suites/test_suite_psa_its.function#L167
will propagate NULL pointer to fwrite
https://github.com/ARMmbed/mbed-crypto/blob/mbedcrypto-1.1.1/library/psa_its_file.c#L212
which must not be NULL.  This is reproduced with appending sanitizer flags -fsanitize=address,undefined flags to CFLAGS in library/Makefile and tests/Makefile and building and running the tests with sanitizers enabled.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

gilles-peskine-arm, 2019-08-06:
```
psa_its_set_wrap( uid, 0, NULL, 0 )
```
ends up calling `psa_its_set( uid, 0, NULL, 0 )`. Since the buffer size is 0, `NULL` is a valid buffer pointer.

The C standard doesn't see it that way (at least for `mem*` functions, NULL/0 is UB; for stdio functions, it doesn't seem to be specified). So we should indeed avoid calling `fwrite` in this case.

Oddly, we _do_ test with Asan and our CI is passing. But I can reproduce the failure locally with gcc5 and `make CFLAGS='-O2 -fsanitize=address,undefined'`. So we should figure out what options we're using, and tweak or add to the CI job.

gilles-peskine-arm, 2019-08-06:
Ok, I understand why CI is passing. I thought we always used Asan and UBsan together (`-fsanitize=address,undefined`), but in `CMakelists.txt`, we activate UBSan with Clang but not with GCC. And we only run that test with ASan on CI through cmake with GCC, so we don't run it with UBSan.

AndrzejKurek, 2019-08-28:
Created a PR here: https://github.com/ARMmbed/mbed-crypto/pull/239

Patater, 2019-09-05:
PR #239 merged with a fix for this.

closed 2019-09-05 by Patater