ARMmbed/mbed-crypto #249: Automate on-target testing

gilles-peskine-arm, 2019-09-10:
### Description

As a developer or maintainer of Mbed Crypto/TLS,
I want `mbedtls_test.py` to be validated by the CI
so that I'm confident that changes to this file work.

Add automation for on-target testing: building the test suites and running them with `mbedtls_test.py`. Run this automation on the CI.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2023-05-24:
`mbedtls_test.py` [no longer exists](https://github.com/Mbed-TLS/mbedtls/pull/5116), so this issue is moot. We plan to design a replacement mechanism, and when we do, ease of use (without a long chain of manual steps) will be a primary consideration.

closed 2023-05-24 by gilles-peskine-arm