ARMmbed/mbed-crypto #250: Support auxiliary code files in the unit test framework

gilles-peskine-arm, 2019-09-10:
### Description

#### Goal

As a developer and maintainer of Mbed TLS/Crypto unit tests,
I want to write auxiliary functions in separate source files which are compiled separately
so as to avoid code duplication, keep file size manageable, and organize the code in a comprehensible way, without increasing compilation time.

At least `helpers.function` should become one such separate file, and probably `host_test.function` and `target_test.function`. These should be ordinary `.c` files since the test script generator doesn't modify them.

#### Rationale

In the test suites, we can't easily write auxiliary functions that are used by more than one `.function` file, or that may or may not be used depending on compile-time options.

To define an auxiliary function that is used by more than one `.function` file, we have two ways:

* Define it in `helpers.function` or `main_test.function`, which are hard-coded in the calls to `generate_test_code.py`.
* Define it in a header file.

This has come up time and again, and is getting worse as we improve our test coverage and thus want more auxiliary functions for fault injection, cleanup, consistency checks, etc. For example, I started to add consistency checks for the PSA crypto API, and I had to define them in a specific header file `tests/psa_crypto_helpers.h`. https://github.com/ARMmbed/mbed-crypto/pull/162 adds generic support for skipping tests when a feature is not supported by an alternative implementation, and adds a lot of boilerplate which could benefit from auxiliary functions. We should be able to define functions in other files, and let the linker link whatever is used.

#### Difficulties

We should change the build scripts for tests to compile `.c` files separately, and then link them together, instead of compiling and linking a single `.c` file at once.

The main difficulty is to update `mbedtls_test.py` and the procedure to use it. This makes https://github.com/ARMmbed/mbed-crypto/issues/249 a de facto prerequisite.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2020-09-10:
Fulfilled by https://github.com/ARMmbed/mbedtls/issues/3325 in Mbed TLS.

closed 2020-09-10 by gilles-peskine-arm