ARMmbed/mbed-crypto #281: Occupy bits in struct psa_key_derivation_s

gilles-peskine-arm, 2019-09-24:
### Description

https://github.com/ARMmbed/mbed-crypto/pull/276 introduced a 1-bit field in `struct psa_key_derivation_s`. We can use 31 more bits for flags or more generally small values that are currently in type-specific parts of the union, e.g. for state data that applies to most KDFs. This would save a tiny amount of RAM (and perhaps code size if the moved stuff is currently at a large offset that requires larger code to access).

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2023-05-24:
With the introduction of the driver interface, this lone bit no longer seems to have potential companions.

closed 2023-05-24 by gilles-peskine-arm