ARMmbed/mbed-crypto #283: Allow passing the public key when registering a key in a secure element

gilles-peskine-arm, 2019-09-26:
### Description

`mbedtls_psa_register_se_key` allows registering a key which is already present in a secure element. The caller passes metadata such as the location, key type and usage policy.

For secure elements that can't export the public key from a private key slot, key registration needs to let the caller pass the public key. Related: #282 for key generation.

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

adbridge, 2019-11-04:
Apologies, this was closed in error by the automated bot - re-opening.

gilles-peskine-arm, 2023-05-24:
The secure element driver interface enabled by `MBEDTLS_PSA_CRYPTO_SE_C` is deprecated. We will not add new features to it.

closed 2023-05-24 by gilles-peskine-arm