ARMmbed/mbed-crypto #319: CMAC with AES_ECB crashes if custom allocator is installed

guidovranken, 2019-11-12:
### Description

If a custom allocator is installed, CMAC with AES_ECB will crash.

```c
#include <stdint.h>
#include <stdlib.h>
#include <stdio.h>
#include <mbedtls/cmac.h>
#include <mbedtls/platform.h>

#define CF_CHECK_EQ(expr, res) if ( (expr) != (res) ) { goto end; }
#define CF_CHECK_NE(expr, res) if ( (expr) == (res) ) { goto end; }

static void* mbedTLS_custom_calloc(size_t A, size_t B) {
    return malloc(A*B);
}

static void mbedTLS_custom_free(void* ptr) {
    free(ptr);
}

int main(void)
{    
    const mbedtls_cipher_info_t *cipher_info = NULL;
    const uint8_t key[32];
    const uint8_t cleartext[16];
    uint8_t out[1024];

    if ( mbedtls_platform_set_calloc_free(mbedTLS_custom_calloc, mbedTLS_custom_free) != 0 ) {
        abort();
    }

    CF_CHECK_NE(cipher_info = mbedtls_cipher_info_from_type(MBEDTLS_CIPHER_AES_256_ECB), NULL);
    CF_CHECK_EQ(mbedtls_cipher_cmac(cipher_info, key, sizeof(key) * 8, cleartext, sizeof(cleartext), out), 0);

end:

    return 0;
}
```

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2019-11-12:
I can't reproduce this crash.
```
# your code in a.c
scripts/config.pl set MBEDTLS_PLATFORM_MEMORY
make
gcc -I include -L library -O -Wall -Wextra -c a.c -lmbedcrypto && ./a.out
```

But I do see a bug in your code: `mbedTLS_custom_calloc` should use `calloc`, not `malloc` (or use `malloc` and then `memset` to 0). The library assumes that the allocator returns memory with a content of all-bits-zero. Does the code work for you if you properly initialize the memory to 0?

guidovranken, 2019-11-12:
Whoops I should have seen that, resolved now, sorry.

closed 2019-11-12 by guidovranken