ARMmbed/mbed-crypto #331: Import EC private key is rejected as key length mis-match

CharleyChu, 2019-12-07:
Call psa_import_key() to import an ASN1 encoding of ECP256R1 private key. It is rejected as following check failed in psa_import_key_into_slot()

  if( PSA_BITS_TO_BYTES( PSA_ECC_CURVE_BITS( curve ) ) != data_length )
        return( PSA_ERROR_INVALID_ARGUMENT );

The expect length is 32 for ECP256R1, but the actually key length (data_lenght) is 33 as a leading 00 is added for the DER representation of positive integers

   [ ] Question
    [ ] Enhancement
    [x ] Bug

gilles-peskine-arm, 2019-12-08:
Hi @CharleyChu,

`psa_import_key` expects the raw private key, not an ASN.1 encoding. For a P256R1 key, the raw private key is always 32 bytes, no matter how many leading zero bits are present.

If you have the key in an ASN.1 encoding, you'll need to convert it. Mbed Crypto can import keys in the format defined in RFC 5915 with `mbedtls_pk_parse_key()`, then you can obtain a `mbedtls_ecp_keypair*` with `mbedtls_pk_ec()` and use `mbedtls_mpi_write_binary(mbedtls_pk_ec(pk)->d, buf, 32)` to extract the private key. We intend to provide more convenient parsing functions in a future version of Mbed Crypto.

closed 2019-12-08 by gilles-peskine-arm

CharleyChu, 2019-12-09:
Thanks Gilles!

This message and any attachments may contain confidential information from Cypress or its subsidiaries. If it has been received in error, please advise the sender and immediately delete this message.