ARMmbed/mbed-crypto #336: Incorrect sizes for secp224k1 keys through the PSA API

gilles-peskine-arm, 2019-12-20:
### Description

The PSA crypto code assumes that a Weierstrass elliptic curve has a “bit size”, that a private key on that curve is that many bits, and a public key is a pair of numbers (coordinates) of that many bits. This is true for most curves, but breaks down for one of the curves that we support: secp224k1, which has 224-bit coordinates (since the order of the field is a 224-bit number) but 225-bit private keys (since the order of the base point is a 225-bit number).

The PSA Crypto specification makes the same incorrect assumption when it describes key formats (documentation of `psa_export_key` and `psa_export_public_key`). Private link: https://github.com/ARMmbed/psa-crypto/issues/352

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

yanesca, 2020-03-16:
@gilles-peskine-arm Is this issue fixed by #330?

gilles-peskine-arm, 2020-03-23:
No, this is not fixed in Mbed Crypto yet.

gilles-peskine-arm, 2023-05-24:
This is tracked in mbedtls at https://github.com/Mbed-TLS/mbedtls/issues/3541

closed 2023-05-24 by gilles-peskine-arm