ARMmbed/mbed-crypto #351: Negative testing for unsupported algorithms

gilles-peskine-arm, 2020-01-22:
### Description

When we disable a feature such as a cryptographic algorithm in `config.h`, our tests generally don't try to invoke the feature at all. We should have some tests that check that when an algorithm is disabled, trying to use it returns the expected “not supported” error (as opposed to crashing, or reporting a success, etc.).

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

yanesca, 2020-03-16:
We don't need to fix this in one go with one big comprehensive solution, small improvements in this direction are also a win.

gilles-peskine-arm, 2020-03-23:
Indeed we don't need to fix everything in one go. However I think we should use a principled, systematic approach where possible. I don't want to end up with manually written test cases with `depends_on:!MBEDTLS_xxx_C` for every single `xxx` and every single library function.

Somewhat related: https://github.com/ARMmbed/mbed-crypto/pull/162 is a systematic approach to allowing crypto functions to return `MBEDTLS_ERR_PLATFORM_FEATURE_UNSUPPORTED` at any time. It doesn't solve this specific issue, however, since it doesn't run `xxx`-related test cases when `MBEDTLS_xxx_C` is disabled.

gilles-peskine-arm, 2023-05-24:
This has been implemented for the PSA API in https://github.com/Mbed-TLS/mbedtls/pull/4076.

Realistically, we aren't going to ever do this for the legacy crypto API.

closed 2023-05-24 by gilles-peskine-arm