ARMmbed/mbed-crypto #356: Doxygen misses "end group" markup (\}) in multiple .h files

tomlogic, 2020-01-31:
https://github.com/ARMmbed/mbed-crypto/blob/819799cfc68e4c4381673a8a27af19802c8263f2/include/mbedtls/ctr_drbg.h#L139

This is just one example, but I noticed Doxygen failing to parse some comments in mbedtls files included in my project, and leaving the "group" open.  I was able to fix this by updating those comments to start with `/**` instead of `/*` as follows:

```
-/* \} name SECTION: Module settings */
+/** \} name SECTION: Module settings */
```

It might be a good idea to look for that pattern and fix it throughout the repository in a single commit.  Unless you're using something other than Doxygen to generate documentation, or perhaps some configuration options that I wasn't using in my doxyconfig.

mpg, 2020-02-04:
Thanks for your report! This looks like a bug indeed, but can you clarify how it shows (in the generated output, or output of the doxygen command) that the end-of-section markers are not closed?

Note: we have a script, `tests/scripts/check-doxy-blocks.pl`, which is precisely intended to "Detect comment blocks that are likely meant to be doxygen blocks but aren't" but is clearly not catching this, we probably want to fix/extend that script as well.

tomlogic, 2020-02-04:
Here's the output of my Doxygen run, with the paths truncated.  That's what turned me on to the problem originally, as I was fixing Doxygen markup in my main project (that includes some of mbedtls).
```
.../include/mbedtls/ctr_drbg.h:381: warning: end of file with unbalanced grouping commands
.../include/mbedtls/entropy.h:290: warning: end of file with unbalanced grouping commands
.../include/mbedtls/platform.h:420: warning: end of file with unbalanced grouping commands
```

mpg, 2020-02-05:
Thanks!

Ok, so there must but some difference between your project's and ours, or our versions of doxygen, because we're not seeing this warning (as evidence by the fact that `tests/scripts/doxygen.sh` passes). (I tried locally with doxygen 1.8.11, and I think our CI is using the same version.)

Anyway, this is a bug and it needs fixing. It would be ideal to know how to reproduce the warning so that we can avoid regressions in the future, but even without that we can fix the bug.

tomlogic, 2020-02-05:
I was seeing the warning with 1.8.16 and 1.8.17.

Here's the project's Doxyfile: https://github.com/digidotcom/xbee_ansic_library/blob/master/Doxyfile

mpg, 2020-02-06:
Thanks for the info! I was able to test with doxygen 1.8.17 and got the same warnings (and several others actually).

For reference, here's the list of unique warnings and their number of occurrences, obtained with `tests/scripts/doxygen.sh | sed -e 's/.*warning://' -e "s/PSA_[^']*/PSA_XXX/" -e 's/psa_[^ ]*/psa_xxx/' | sort | uniq -c`:

* in mbed-crypto:
```
      1  Tag 'MSCGEN_PATH' at line 1707 of file 'mbedtls.doxyfile' has become obsolete.
      1  Tag 'PERL_PATH' at line 1686 of file 'mbedtls.doxyfile' has become obsolete.
      1  end of file while inside a group
     10  end of file with unbalanced grouping commands
     48  return value '#PSA_XXX' of psa_xxx has multiple documentation sections
```
* in mbedtls:
```
      1  Refusing to add group x509_module to itself
      1  Tag 'MSCGEN_PATH' at line 1713 of file 'mbedtls.doxyfile' has become obsolete.
      1  Tag 'PERL_PATH' at line 1692 of file 'mbedtls.doxyfile' has become obsolete.
      5  end of file while inside a group
     13  end of file with unbalanced grouping commands
     48  return value '#PSA_XXX' of psa_xxx has multiple documentation sections
```

I suggest we keep this issue about the grouping warnings in mbed-crypto ("end of file while inside a group" and "end of file with unbalanced grouping commands"), and I'll open other issues for the other types of warnings.

Thanks again for reporting the issue and providing information allowing us to reproduce it!

mpg, 2020-07-20:
Let's continue this in ARMmbed/mbedtls#3497

closed 2020-07-20 by mpg