ARMmbed/mbed-crypto #359: Recent doxygen throws warnings "return value XXX of xxx has multiple documentation sections"

mpg, 2020-02-06:
### Description

When running `tests/scripts/doxygen.sh` with doxygen 1.8.17, the script fails with a bunch of warnings, in particular there are 48 warnings looking like
```
warning: return value '#PSA_XXX' of psa_xxx has multiple documentation sections
```

This issue is about those warnings (other warnings thrown by doxygen 1.8.17 are tracked in other issues: #356 and [TODO]).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2020-02-06:
This was deliberate: I used `\retval FOO` twice if there were two error conditions that led to the same return value. There's nothing in the [documentation](http://www.doxygen.nl/manual/commands.html#cmdretval) to say that this isn't the normal way to use it :frowning_face:

mpg, 2020-02-06:
If this is deliberate, perhaps there's a way to turn the warning off in the doxyfile?

gilles-peskine-arm, 2020-07-20:
Let's continue this in https://github.com/ARMmbed/mbedtls/issues/3497

closed 2020-07-20 by gilles-peskine-arm