ARMmbed/mbed-crypto #360: Doxygen 1.8.17 throws warnings about our doxyfile

mpg, 2020-02-06:
### Description

When running tests/scripts/doxygen.sh with doxygen 1.8.17, the script fails with a bunch of warnings, in particular there are 2 warnings about our doxyfile:

```
Tag 'MSCGEN_PATH' at line 1713 of file 'mbedtls.doxyfile' has become obsolete.
Tag 'PERL_PATH' at line 1692 of file 'mbedtls.doxyfile' has become obsolete.
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2020-02-06:
Our doxyfile is the default doxyfile from some past doxygen version, with some options modified to suit our project. We should replace that by a doxyfile that only mentions the options that we want to modify from the default.

mpg, 2020-07-20:
Let's continue this in ARMmbed/mbedtls#3497

closed 2020-07-20 by mpg