ARMmbed/mbed-crypto #380: 30% of PSA Cryptographic API tests are failed

AndreyButokNXP, 2020-03-02:
**The 23 tests of the PSA Cryptographic Test Suite** ( https://github.com/ARM-software/psa-arch-tests) **are failed** with the latest MbedCrypto.
The main reason of test failures is unimplemented PSA functions by mbedCrypto. 
**More info is here: https://github.com/ARM-software/psa-arch-tests/blob/master/api-tests/docs/test_failure_analysis.md**
The test log is used for PSA Certification. Do you have any plan/dates to fix it?

All targets. All compilers. The latest PSA-Arch_Tests.

    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2020-03-02:
@AndreyButokNXP Thanks for the notice. There are many different things here.

* We've just added `psa_hash_compute`, `psa_hash_compare`, `psa_mac_compute` and `psa_mac_verify`, so they'll be in TF-M once TF-M updates their crypto code.
* `psa_open_key` and `psa_close_key` are present in Mbed Crypto, which implements version 1.0beta3 of the specification, but they have been removed from version 1.0 final of the specification. I think we'll update Mbed Crypto in 2020Q2.
* For the other unimplemented functions: we will implement them, but I can't give you a date.
* “psa_asymmetric_encrypt does not have support for ECC keys” — that's true, the specification currently does not define any algorithm for `psa_asymmetric_encrypt` that uses ECC keys. What's the problem there?
* For the incorrect key derivation error codes, what are the problematic inputs?
* For “psa_generate_key generates incorrect key length for RSA”, what are the problematic inputs?

Please note that the Mbed TLS team does not currently monitor test results from the generic PSA test suite, which target a slightly different version of the API. We have our own tests. If the generic PSA test suite fails and it's not due to an API version difference or a bug in the test suite, please file an issue with details on how to reproduce the problem.

AndreyButokNXP, 2020-03-02:
From Soby Mathew (ARM, TFM project):

> 1)sa_asymmetric_encrypt does not have support for ECC keys
> that's true, the specification currently does not define any algorithm 
> for psa_asymmetric_encryptï¿½that uses ECC keys. What's the problem there?

The PSA-ACK test need to fix this. I will highlight this issue to them.

> 2) For the incorrect key derivation error codes, what are the 
> problematic inputs?

There is an issue raised with mbed-crypto team discussing this issue here : https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FARMmbed%2Fmbed-crypto%2Fissues%2F175&amp;data=02%7C01%7Candrey.butok%40nxp.com%7C53dbcc21b5b549021a3908d7bebad35a%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C1%7C637187581930659738&amp;sdata=4gFvjPJ5sQ0icY1K6vLaTTiZig11S68%2FWolkoy2rcxg%3D&amp;reserved=0

As I understand, this needs to be fixed by mbed-crypto.

> 
> 3) For psa_generate_key generates incorrect key length for RSA
> what are the problematic inputs?
> 
> Could you clarify or this is the PSA-Test-Suite task?

The problematic input can be seen here : 
https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FARM-software%2Fpsa-arch-tests%2Fblob%2Fmaster%2Fapi-tests%2Fdev_apis%2Fcrypto%2Ftest_c016%2Ftest_data.h%23L78&amp;data=02%7C01%7Candrey.butok%40nxp.com%7C53dbcc21b5b549021a3908d7bebad35a%7C686ea1d3bc2b4c6fa92cd99c5c301635%7C0%7C1%7C637187581930659738&amp;sdata=%2B2ilDbDeb1MbCSlY%2BgsIGq3qTJix0dXxy5ChBHZee1s%3D&amp;reserved=0

Please contact Soby Mathew if more clarification is needed.

gilles-peskine-arm, 2020-03-02:
Thank you. The RSA one is [a bug in the test suite](https://github.com/ARM-software/psa-arch-tests/issues/155). We're tracking the others, so they're on our backlog. Please keep reporting any specific bugs!

jaypit02, 2020-03-03:
Raised tracking issue for the missing PSA Crypto APIs implementation - https://github.com/ARMmbed/mbed-crypto/issues/381

gilles-peskine-arm, 2020-03-09:
We have issues for each category of failures. I believe this issue is therefore redundant, so I am closing it. If this is a mistake, please let me know and I'll reopen this issue or raise missing issues.

You can track issues related to the compliance of Mbed Crypto with the PSA Crypto API by looking at [issues labeled "compliance"](https://github.com/ARMmbed/mbed-crypto/labels/compliance).

closed 2020-03-09 by gilles-peskine-arm