ARMmbed/mbed-crypto #381: Few of the Crypto APIs of PSA Crypto 1.0 Beta-3 are implemented mbedcrypto-3.0.1

jaypit02, 2020-03-03:
When running the PSA Crypto APIs tests from https://github.com/ARM-software/psa-arch-tests (Hash: e44ccebdff9631e2474d85650c1812d0ce001dc3) with TFM (Hash: 6e7be077eabef00b4825e245604148d82b11f832), they are tests failures due to missing implementation below APIs in mbedcrypto (https://github.com/ARMmbed/mbed-crypto.git -b mbedcrypto-3.0.1)

psa_hash_compute
psa_hash_compare
psa_aead_encrypt_setup
psa_aead_decrypt_setup
psa_aead_generate_nonce
psa_aead_set_nonce
psa_aead_set_lengths
psa_aead_update_ad
psa_aead_update
psa_aead_finish
psa_aead_verify
psa_aead_abort
psa_mac_compute
psa_mac_verify
psa_cipher_encrypt
psa_cipher_decrypt

Please guide us with plan for supporting these APIs.

gilles-peskine-arm, 2020-03-03:
`psa_hash_compute`, `psa_hash_compare`, `psa_mac_compute` and `psa_mac_verify` are implemented since Mbed Crypto 3.1.0.

gilles-peskine-arm, 2020-04-27:
* `psa_hash_compute`, `psa_hash_compare`: done in #327.
* `psa_aead_encrypt_setup`, `psa_aead_decrypt_setup`, `psa_aead_generate_nonce`, `psa_aead_set_nonce`, `psa_aead_set_lengths`, `psa_aead_update_ad`, `psa_aead_update`, `psa_aead_finish`, `psa_aead_verify`, `psa_aead_abort`: tracked in https://github.com/ARMmbed/mbedtls/issues/3257.
* `psa_mac_compute`, `psa_mac_verify`, `psa_cipher_encrypt`, `psa_cipher_decrypt`: tracked in https://github.com/ARMmbed/mbedtls/issues/3258.

You can now see our planning for these issues on our [sprint board](https://github.com/ARMmbed/mbedtls/projects/3), so I'm closing this issue here.

closed 2020-04-27 by gilles-peskine-arm