ARMmbed/mbed-crypto #382: mbedcrypto-3.0.1 doesn't have the support for persistent keys

jaypit02, 2020-03-03:
When running the PSA Crypto APIs tests from https://github.com/ARM-software/psa-arch-tests (Hash: e44ccebdff9631e2474d85650c1812d0ce001dc3) with TFM (Hash: 6e7be077eabef00b4825e245604148d82b11f832), there are tests failures due to missing implementation below APIs in mbedcrypto (https://github.com/ARMmbed/mbed-crypto.git -b mbedcrypto-3.0.1)

The implementation of psa_open_key/psa_close_key requires the support for persistent key storage. Since mbedcrypto doesn't have support for storing persistent keys, psa_import_key fails for the same(Error: -134 - UNSUPPORTED).

Test failures (Target- AN521):
TEST: 250 | DESCRIPTION: Testing crypto key management APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_open_key 16 Byte AES
        Failed at Checkpoint: 6
        Actual: -134
        Expected: 0
TEST RESULT: FAILED (Error Code=0x00000001) 

******************************************

TEST: 251 | DESCRIPTION: Testing crypto key management APIs
[Info] Executing tests from non-secure
[Check 1] Test psa_close_key 16 Byte AES
[Check 2] Test psa_close_key 24 Byte AES
        Failed at Checkpoint: 3
        Actual: -134
        Expected: 0
TEST RESULT: FAILED (Error Code=0x00000001) 

Please guide us on planning the support for persistent key storage.

gilles-peskine-arm, 2020-03-03:
Mbed Crypto does have persistent key storage. It works on top of the PSA ITS API. Mbed Crypto provides an implementation of PSA ITS over standard C stdio. It can also work with other PSA ITS implementations.

I can't speak to the TF-M integration.

closed 2020-03-03 by gilles-peskine-arm

jaypit02, 2020-03-03:
@soby-mathew has confirmed that this is a TFM issue. Therefore, will raise TFM issue.