ARMmbed/mbed-crypto #385: i386 builds fail with GCC7

sbutcher-arm, 2020-03-04:
### Description

Building with GCC7, the `all.sh` test `test_m32_everest` component fails with the following warning:

```
suites/test_suite_psa_crypto.function: In function 'test_generate_random':
suites/helpers.function:398:25: error: argument 2 value '4294967277' exceeds maximum object size 2147483647 [-Werror=alloc-size-larger-than=]
             ( pointer ) = mbedtls_calloc( sizeof( *( pointer ) ), \
             ~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                           ( length ) );           \
                                           ~~~~~~~~~~~~
suites/test_suite_psa_crypto.function:5301:5: note: in expansion of macro 'ASSERT_ALLOC'
In file included from suites/helpers.function:5:0:
/usr/include/stdlib.h:541:14: note: in a call to allocation function 'calloc' declared here
 extern void *calloc (size_t __nmemb, size_t __size)
              ^~~~~~
```

In the test suite `test_suite_psa` there's a test case function of `generate_random()` which takes `int` as a parameter `bytes_arg` and assigns it to `bytes` as a `size_t`. The parameter comes from the data file so comes in via standard IO so could be at the limit. Within this function, it makes a call to the macro `ASSERT_ALLOC()`, passing that parameter `bytes` + `sizeof(trail)`, without any boundary checks on `bytes` unchecked.

Within `ASSERT_ALLOC` there is then a call to `calloc` that takes a parameter of size as `size_t`, and passes `bytes_arg` an `int` into it.

In the Everest m32 test, this is failing with the warning? Why? Well, we've moved from LP64, where `int` is 32bits, and `size_t` is 64 bits, to ILP32 where `int` is 32bits and `size_t` is 32 bits, leading to a potential underflow.

This issue already has a PR to fix it with PR #377.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

gilles-peskine-arm, 2020-03-04:
More precisely, GCC 7 emits this warning when building with UBSan (`-fsanitize=undefined`) and `-O2`. It looks like a GCC bug where some transformations caused by an optimization such as inlining causes additional warnings that are not relevant to the original code. See https://github.com/ARMmbed/mbed-crypto/pull/377#issuecomment-593364500

GCC bug or not, we do want our code to build without warnings on all non-ancient GCC versions.

gilles-peskine-arm, 2020-09-10:
This was fixed back in https://github.com/ARMmbed/mbed-crypto/pull/377, we just forgot to close the issue.

closed 2020-09-10 by gilles-peskine-arm