ARMmbed/mbed-crypto #389: Don't use FindPythonInterp

okhowang, 2020-03-18:
### Description

### Issue request type

    [ ] Question
    [x] Enhancement
    [ ] Bug

see [https://cmake.org/cmake/help/latest/module/FindPythonInterp.html](https://cmake.org/cmake/help/latest/module/FindPythonInterp.html)
FindPythonInterp may failed when python 2 found first

gilles-peskine-arm, 2020-03-23:
Note that we do want to maintain backward compatibility with somewhat older versions of cmake, at least cmake 3.5.1 which is the version of Ubuntu 16.04. So we can't just switch to `FindPython3` or `FindPython`.

okhowang, 2020-03-24:
can we add a progressive upgraded behaviour?
for example if cmake version > 3.x do `FindPython3` instead of `FindPythonInterp`

gilles-peskine-arm, 2020-03-24:
Sure. As long as the file can be parsed with older versions, that's ok. We have tests on `GCC_VERSION`, we can have tests on `CMAKE_VERSION` as well (or on `cmake_is_module_available("FindPython3")` if there's such a thing?).

closed 2020-06-10 by danh-arm